ocrdu/Arduino_SAMD21_turbo_PWM #2: Appears to work with Adafruit ItsyBitsy Express M0

jstanle1, 2020-04-04:
Thanks for your work on this library -- it's saved me a lot of trouble and, as you put it, messing with dark corners of the SAMD21 spec.

Just hacked the header file to allow it to accept Adafruit ItsyBitsy Express M0 board and it works at least for a couple of pins (which is all I need right now); not an exhaustive test, but working well for what I need now.

ocrdu, 2020-07-27:
Hi,

Thanks for checking it out on different hardware. Did you publish the changed header somewhere?

jstanle1, 2020-07-27:
No, I never published the change.  Here’s what I added around line 69 of the original header:

 

#elif defined (ARDUINO_SAMD_ZERO) || \

      defined (ARDUINO_SAMD_FEATHER_M0) || \

      defined (ADAFRUIT_ITSYBITSY_M0)

 

Just enabling it for the ItsyBitsy board was all it took; it works well.

 

Thanks for publishing your code – it saved me a lot of time and effort.

 

Regards,

  Jim Stanley

ocrdu, 2020-07-27:
Looks like an easy change. I'll check all pin mappings when I can find the time.

ocrdu, 2020-07-27:
h

closed 2020-08-20 by ocrdu