ocrdu/Arduino_SAMD21_turbo_PWM #3: "uninitialized const member 'PinLookup::*'" errors

alexchumak, 2020-06-24:
I had to remove the "const" prefix from the definition in the header file for it to compile. Am I missing something?

```
~/Documents/Arduino/libraries/SAMD21_turbo_PWM/src/SAMD21turboPWM.h:112:1: error: uninitialized const member 'PinLookup::port'
 };
 ^
~/Documents/Arduino/libraries/SAMD21_turbo_PWM/src/SAMD21turboPWM.h:112:1: error: uninitialized const member 'PinLookup::samd21Pin'
~/Documents/Arduino/libraries/SAMD21_turbo_PWM/src/SAMD21turboPWM.h:112:1: error: uninitialized const member 'PinLookup::timer'
~/Documents/Arduino/libraries/SAMD21_turbo_PWM/src/SAMD21turboPWM.h:112:1: error: uninitialized const member 'PinLookup::pMux'
~/Documents/Arduino/libraries/SAMD21_turbo_PWM/src/SAMD21turboPWM.h:112:1: error: uninitialized const member 'PinLookup::port'
~/Documents/Arduino/libraries/SAMD21_turbo_PWM/src/SAMD21turboPWM.h:112:1: error: uninitialized const member 'PinLookup::samd21Pin'
~/Documents/Arduino/libraries/SAMD21_turbo_PWM/src/SAMD21turboPWM.h:112:1: error: uninitialized const member 'PinLookup::timer'
~/Documents/Arduino/libraries/SAMD21_turbo_PWM/src/SAMD21turboPWM.h:112:1: error: uninitialized const member 'PinLookup::pMux'
```

ocrdu, 2020-07-27:
Hi,

I don't know why that happens; many parts of C++ are dark an dangerous to me. Only thing I can think of is that your compiler is set to a more pedantic mode than mine. I haven't heard about this from other users (yet).

I don't initialise everything in the lookup tables in the header file, only what is used; maybe that is the problem, combined with compiler settings.

I t does work properly for you, after your fix?

jwestmoreland, 2020-08-30:
Hello @ocrdu ,

Actually, in your PinLookup struct - arduinoPin should be declared:

const int arduinoPin;        // Arduino pin number

since you're using -1's as initializers for undefined lines.

That cleared it up for me - otherwise I couldn't compile.

Thanks,
John W.

ocrdu, 2020-08-30:
Thanks! I fixed that, and got rid of some compiler warnings as well, in version 0.8.0. Let me know if that has worked for you.

I wonder why 0.7.0 it did compile for me and some others; consider this I must.

jwestmoreland, 2020-08-30:
Hello @ocrdu ,

Version 0.8.0 built fine for me.

Which version of the tools are you building with?

This is the latest I have from Arduino:

\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\7-2017q4/bin/arm-none-eabi-g++

Thanks,
John W.

ocrdu, 2020-08-30:
Hi @jwestmoreland,

That is the same one I have and use. I would have expected the compiler to yell at me and call me names, but no. Oh well, it's fixed now.

ocrdu, 2020-08-30:
Hi @alexchumak,

I could reproduce the errors you show as warnings only, which is a bit surprising, but I think I have fixed things in version 0.8.0. Please let me know if that works for you.

jwestmoreland, 2020-08-30:
Hello @ocrdu ,

You can promote warnings to errors in compiler directives; but I don't know how you'd demote an error to a warning - that's typically not possible - at least I think (unless you're running an experimental version of a compiler).

Just so you know - I'm using your lib over [here:](https://github.com/jwestmoreland/LedDisplay)

Thanks,
John W.

ocrdu, 2020-08-31:
Hi @jwestmoreland,

Can't test your code for want of that display, but it should be able to make smoother transistions in brightness, especially if you want to "pulse" the display.

jwestmoreland, 2020-09-01:
Hello @ocrdu ,

Tested and works OK - definitely gives more control over brightness than the default method (setting 4 bit value).

Thanks!
John W.

closed 2020-09-01 by ocrdu