ocrdu/Arduino_SAMD21_turbo_PWM #5: Different PWM Pins

dakrulski, 2021-02-21:
Hi,
thank you very much for this library :slightly_smiling_face: 
I would like to ask you if it would be possible to choose the original Arduino nano 33 IoT PWM pins (2, 3, 5, 6, 9, 10, 11, 12) ?
Is it possible to change the `pinTable` in `src/SAMD21turboPWM.h` to match the original PWM pins?
The Arduino nano 33 IoT is still a bit new to me and I'm afraid I might break something or brick the Arduino if I change the values. Are there things I should watch out for?

ocrdu, 2021-02-22:
There really isn't such a thing as a PWM pin on the SAMD21; they can be configured in software, within some limitations and restrictions. I never considered the "original" PWM pins when writing the library.

It is possible to change the pinTable, but not all combinations of settings will work, see https://github.com/ocrdu/Arduino_SAMD21_turbo_PWM/blob/master/extras/pintables.txt for which will. 

I doubt you could brick the 33 IoT when you try, but I wouldn't go there, and just use the pins as provided in the library.

closed 2021-02-24 by ocrdu

dakrulski, 2021-02-28:
Thank you very much for your answer. I totally missed the `extras/pintables.txt`. I though about it and I fixed my PCB to match your pins and everything works fine. :+1:
Thanks again for this great library :slightly_smiling_face:

silvan2468, 2022-04-23:
Hy ocrdu
Thanks also from me for this library.
But I hope you will think for the arduino nano 33 iot about the pins you use. Because they are completely different than what the original arduino pinout shows. Also with the original pinout will be 11 pwm pins possible and not "only" 8. From which 3 are occupied by the SPI pins.
I hope you will change the pin table for the nano iot or perhaps doing a second table named "iot ORIGINAL"?
So that everyone can commen/uncoment in the .h-File which table he likes...

Thanks and kind regards
Silvan

silvan2468, 2022-04-23:
I added it myself. Here for thowse who likes it the "original way" ;)

{-1, 0, 0, 0, 0, 0},
{-1, 0, 0, 0, 0, 0},
{-1, 0, 0, 0, 0, 0},
{-1, 0, 0, 0, 0, 0},
{-1, 0, 0, 0, 0, 0},
{ 5, PORTA,  5, 0, &REG_TCC0_CCB1, PORT_PMUX_PMUXO_E},
{ 6, PORTA,  4, 0, &REG_TCC0_CCB0, PORT_PMUX_PMUXE_E},
{-1, 0, 0, 0, 0, 0},
{-1, 0, 0, 0, 0, 0},
{ 9, PORTA, 20, 0, &REG_TCC0_CCB2, PORT_PMUX_PMUXE_F},
{10, PORTA, 21, 0, &REG_TCC0_CCB3, PORT_PMUX_PMUXO_F},
{11, PORTA, 16, 2, &REG_TCC2_CCB0, PORT_PMUX_PMUXE_E},
{-1, 0, 0, 0, 0, 0},
{13, PORTA, 17, 2, &REG_TCC2_CCB1, PORT_PMUX_PMUXO_E},
{-1, 0, 0, 0, 0, 0},
{-1, 0, 0, 0, 0, 0},
{16, PORTA, 11, 1, &REG_TCC1_CCB1, PORT_PMUX_PMUXO_E},
{17, PORTA, 10, 1, &REG_TCC1_CCB0, PORT_PMUX_PMUXE_E} 

Or do you like to implement it?

What I would more prefer, if you could please add timer 5. So then with the arduino nano 33 iot there would be two more pwm pins (pins 2+3).

Thanks a lot.
Silvan